knee-koh/MarIOnette #3: Installing on Linux

baldyman01, 2023-04-12:
Blender 3.5 on Ubuntu 21.04

When enabling the Addon in Blender 3.5 on Ubuntu it throws the following error -

_`Traceback (most recent call last):
  File "/home/pete/.config/blender/3.5/scripts/addons/MarIOnetteRC1.py", line 19, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pete/Applications/Blender_Launcher/stable/blender-3.5.0+stable.1be25cfff18b/3.5/scripts/modules/addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "/home/pete/.config/blender/3.5/scripts/addons/MarIOnetteRC1.py", line 25, in <module>
    if platform.system() == 'Windows':
NameError: name 'platform' is not defined
`_

This is caused by pyserial module being missing from the Blender python install.

To fix this -

Fix for No module named serial -
  
Get the get-pip.py file from the pip documentation - https://pip.pypa.io/en/stable/installation/

You'll find the blender python binary at: /blender-path/3.xx/python/bin/python3.10

Use this binary to run the get-pip.py. If you've saved it in your home directory the command to use should look something like this:

$ /blender-path/3.xx/python/bin/python3.10 ~/get-pip.py

[image]

You should now have pip installed for blender. You use it with blenders python and you have to point to the pip that was installed for blenders python. Both are in blenders folder tree and you use them like this:

$ /blender-path/3.xx/python/bin/python3.10 /blender-path/3.xx/python/lib/python3.5/site-packages/pip install pyserial
 
[image]

Restart Blender and it Enables with no errors.

knee-koh, 2023-04-12:
Thanks for looking into this! I just pushed a new release v1_0_1, that should install on Linux without having to open the terminal. I tested on Ubuntu 22.04 LTS. Give the new release a try to see if it works.

baldyman01, 2023-04-13:
Thanks. Now fixed on 20.04.

Cheers

Pete

closed 2023-04-13 by knee-koh

stb1Cluny, 2025-01-25:
Same problem on Ubuntu 24.04. Unfortunately it's not so easy to get pip to install to Blender's site-packages. 
I got a warning running pip per @baldyman01's solution:
`Defaulting to user installation because normal site-packages is not writeable`
^I thought this might be how Blender is installed by Snap on Ubuntu so I also tried downloading the .tar direct from Blender. Same issue.
I'll add a another comment in the newer issue here: https://github.com/knee-koh/MarIOnette/issues/21