knee-koh/MarIOnette #6: was not declared in this scope?????

Shaorune777, 2023-07-07:
hello. 
I kinda new with this type of code. usually I just run with .h file when I want to animate simple thing.

so the problem I currently I got from this is. I cant compile it in IDE

> C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:81:32: error: 'TOTAL_MOTORS' was not declared in this scope
 const int expectedMotorBytes = TOTAL_MOTORS * 2;
                                ^~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino: In function 'void readSerialBytes()':
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:164:14: error: 'TOTAL_MOTORS' was not declared in this scope
           if(TOTAL_MOTORS > 0){
              ^~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:172:18: error: 'motor_values' was not declared in this scope
               if(motor_values[i][0] == 1){
                  ^~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:172:18: note: suggested alternative: 'servo2_value'
               if(motor_values[i][0] == 1){
                  ^~~~~~~~~~~~
                  servo2_value
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:191:14: error: 'TOTAL_LEDS' was not declared in this scope
           if(TOTAL_LEDS > 0){
              ^~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:196:18: error: 'led_values' was not declared in this scope
               if(led_values[i][0] == 11){
                  ^~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:196:18: note: suggested alternative: 'ledValue_B'
               if(led_values[i][0] == 11){
                  ^~~~~~~~~~
                  ledValue_B
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:202:18: error: 'led_values' was not declared in this scope
               if(led_values[i][0] == 10 && tempBuffer[offset] == 1){
                  ^~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:202:18: note: suggested alternative: 'ledValue_B'
               if(led_values[i][0] == 10 && tempBuffer[offset] == 1){
                  ^~~~~~~~~~
                  ledValue_B
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino: In function 'void setupLEDs()':
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:263:6: error: 'TOTAL_PWM_LEDS' was not declared in this scope
   if(TOTAL_PWM_LEDS > 0){
      ^~~~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:264:24: error: 'TOTAL_LEDS' was not declared in this scope
     for(int i = 0; i < TOTAL_LEDS; i++){
                        ^~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:265:10: error: 'led_values' was not declared in this scope
       if(led_values[i][0] == 11){
          ^~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:265:10: note: suggested alternative: 'ledValue_B'
       if(led_values[i][0] == 11){
          ^~~~~~~~~~
          ledValue_B
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:272:6: error: 'TOTAL_NEOPIXELS' was not declared in this scope
   if(TOTAL_NEOPIXELS > 0){
      ^~~~~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:273:24: error: 'TOTAL_LEDS' was not declared in this scope
     for(int i = 0; i < TOTAL_LEDS; i++){
                        ^~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:274:10: error: 'led_values' was not declared in this scope
       if(led_values[i][0] == 10){
          ^~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:274:10: note: suggested alternative: 'ledValue_B'
       if(led_values[i][0] == 10){
          ^~~~~~~~~~
          ledValue_B
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino: In function 'void setupMotors()':
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:318:22: error: 'TOTAL_MOTORS' was not declared in this scope
   for(int i = 0; i < TOTAL_MOTORS; i++){
                      ^~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:320:8: error: 'motor_values' was not declared in this scope
     if(motor_values[i][0] == 1){
        ^~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:320:8: note: suggested alternative: 'servo2_value'
     if(motor_values[i][0] == 1){
        ^~~~~~~~~~~~
        servo2_value
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino: In function 'void setupSteppers()':
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:332:22: error: 'TOTAL_STEPPERS' was not declared in this scope
   for(int i = 0; i < TOTAL_STEPPERS; i++){
                      ^~~~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:333:24: error: 'TOTAL_MOTORS' was not declared in this scope
     for(int j = 0; j < TOTAL_MOTORS; j++){
                        ^~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:334:10: error: 'motor_values' was not declared in this scope
       if(motor_values[j][0] == 5){
          ^~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:334:10: note: suggested alternative: 'servo2_value'
       if(motor_values[j][0] == 5){
          ^~~~~~~~~~~~
          servo2_value
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino: In function 'void setup()':
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:366:6: error: 'TOTAL_MOTORS' was not declared in this scope
   if(TOTAL_MOTORS > 0){
      ^~~~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:372:6: error: 'TOTAL_LEDS' was not declared in this scope
   if(TOTAL_LEDS > 0){
      ^~~~~~~~~~
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino: In function 'void loop()':
C:\Users\Regan\Downloads\MarIOnette-main\Arduino Code\MarIOnette_Template_V1\MarIOnette_Template_V1.ino:390:6: error: 'TOTAL_STEPPERS' was not declared in this scope
   if(TOTAL_STEPPERS > 0){
      ^~~~~~~~~~~~~~

exit status 1

Compilation error: 'TOTAL_MOTORS' was not declared in this scope 
``

I jsut dont know what to do. 
I already download all the library. is there Something I missing???
btw I use arduino uno board.

knee-koh, 2023-07-07:
Hi, does your Arduino Sketch folder include the config.h file? This gets generated by the Blender MarIOnette script and includes the variables that are shown in your error message.

Shaorune777, 2023-07-08:
oh yeah. before this error occurring. I got error that need config.h file. 
but I cant find the config.h 
so I just download it. called GRBL-Arduino-library-master.
after that I installed. and compile it again. 
and here its the error I got.

but I dont find any config.h file in the Arduino script folder. is there I missing something????

closed 2023-07-11 by knee-koh