knee-koh/MarIOnette #7: Cannot install on Blender 3.6 on  Ubuntu

baldyman01, 2023-08-12:
Linux MONSTER 5.15.0-78-generic #85~20.04.1-Ubuntu SMP Mon Jul 17 09:42:39 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
Not working-  3.6.1 , 3.6
Working  - 3.5.1

Installing MarIONette 1.0.2 gives me a normal looking return in the terminal - 

`Modules Installed () from '/home/pete/Downloads/MarIOnette-1.0.2.zip' into '/home/pete/.config/blender/3.6/scripts/addons'
`
However the addon does not appear in the preferences->addon tab

Files have been copied as indicated on the log but starting Blender from terminal throws no errors at all. The addon just appears to be silently ignored.

 Blender 3.5 works as expected.

knee-koh, 2023-08-15:
Hi, thank you for opening the issue! I have read that there have been some API changes for Blender 3.6, which affect the way addons are installed + run. Unfortunately I will not be able to test this right away, so running MarIOnette in an older version like you tested (3.5 or earlier) is likely the best solution until I can address this.

baldyman01, 2023-08-26:
OK, tried this again after a reboot. It works OK now. No idea what that was about.

closed 2023-08-26 by baldyman01