knee-koh/MarIOnette #8: Can't Connect Blender to Arduino

iegl3, 2023-08-15:
Python: Traceback (most recent call last):
  File "C:\Users\Us\AppData\Roaming\Blender Foundation\Blender\3.6\scripts\addons\MarIOnette v1_0_2.py", line 1253, in execute
    serialport = serial.Serial(scn.scn_prop.PortSelect, scn.scn_prop.BaudRate, timeout = 0.01)
  File "C:\Users\Us\AppData\Roaming\Python\Python310\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\Us\AppData\Roaming\Python\Python310\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Users\Us\AppData\Roaming\Python\Python310\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM5': PermissionError(13, 'Access is denied.', None, 5)

knee-koh, 2023-08-15:
Hi, are you able to open this COM port in Arduino IDE?

I have not yet been able to test on Blender 3.6, as there have been some API changes that need to be addressed. The Blender version you are using could be the problem, but I cannot confirm this until I am able to test myself.

knee-koh, 2023-12-09:
I've been able to test on blender versions up to 4.02 on Windows 10 and Mac M1, and have not seen this issue come up. Could you please update to MarIOnette 1.0.3 and re-install the add-on (opening Blender in Administrator mode if on Windows) and see if you still have problems?

AustinRamby75, 2024-08-04:
I am having a similar issue. Just downloaded today. When Im in my initial setup, it can find the COM port just fine. However, when I synch and add it to my arduino, there are no options for Serial Port in Blender.
[image]

knee-koh, 2024-08-23:
Are you getting the same error message as the first user? Your issue sounds like a Serial Port problem:
- Which Arduino are you using?
- How many serial ports does that Arduino have?
- Do you have a serial monitor open in the Arduino IDE, which could be blocking MarIOnette from seeing it?

closed 2025-01-02 by knee-koh

MLRigdon, 2025-05-22:
I am having the same issue as the first person. Could it be due to using a servo motor driver?