knee-koh/MarIOnette #10: Blender starts lagging after bones move

longarmlieshere, 2023-12-08:
I'm facing this issue where after (roughly) 9 seconds of constant bone rotation, Blender lags to the point where it's unusable. Disconnecting fixes the lag. I'm using MarIOnette v1.0.3 and Blender 4.0.1, but have also tried a fresh install of 3.4.1. I'm under the assumption this has something to do with the values, as it won't lag if you just only connect to the serial port. Bones have to be moving for the lag to start.

[Here is a video demonstrating the issue.](https://www.youtube.com/watch?v=yryrCTdEgC0) I don't know what info is or isn't important in this case, so if there's anything I left out just let me know. Thank you.

knee-koh, 2023-12-08:
Thanks for submitting the issue and the video! It's really helpful for debugging. I've run 40+ actuators and LEDs for over 5 minutes at a time, so this sounds like it might be a hardware (microcontroller) or setup issue.

Let's work through some common problems first. I've seen Blender+MarIOnette lock up and freeze like this when the wrong port is selected and connected to. 

What kind of microcontroller are you using (and does the port number match the one found in arduino app)?

longarmlieshere, 2023-12-09:
Thanks for replying so fast! I'm using an Arduino Uno R3 with a Baud Rate of 19200. Every time I plug it into my computer it shows up as "COM4", so I've just assumed it's that one.

knee-koh, 2023-12-09:
That's helpful to know, thank you. Do you mind sharing the .blend, .ino, and .h files? I'll try to replicate the issue on my end to see what's happening in more detail.

longarmlieshere, 2023-12-09:
Sure, here you go: [Servo.zip](https://github.com/knee-koh/MarIOnette/files/13622340/Servo.zip)

knee-koh, 2023-12-09:
So strange... I just tested on both an M1 Mac and a Windows 10 machine, and am able to connect to the Uno R3 and drive a servo on pin 3 for longer than 9 seconds (I got bored after 30 seconds, haha).

- Does the servo move before Blender is frozen?
- Is the Arduino Uno R3 a genuine one, or a clone? I had some issues with Arduino clones connecting and programming, which could mean the Serial connection could give out
- Another thing to try: try increasing the baud rate to 115200, hitting the sync button, and re-uploading the sketch to the Arduino
- Finally, you can try increasing the gap between serial reads: on line 28 of the .h file, change the SERIAL_DELAY value from 1 to 5, then re-upload and see if that makes a difference

Sorry this is giving you issues! I hope we can fix it so you can get up and running ASAP.

longarmlieshere, 2023-12-11:
Don't be sorry, it's all good! I appreciate you taking the time to help me out with this.

- Servos do move, but once Blender starts lagging they stop completely.
- Pretty sure it's genuine, bought it off of Amazon not long ago.
- 115200 didn't seem to do anything.
- This one actually worked for a bit, but started lagging after about 24 seconds.

I'm gonna try using a spare computer I have to see what happens shortly

knee-koh, 2023-12-12:
Great, I hope the spare computer will work out!

Some other things that I've seen affect Serial performance/connectivity:

- If you're using a usb hub, try connecting directly to a port on your machine
- Try a different USB cable if you haven't already
- Try a different servo; the reason I bring this up is sometimes Servos can pull a lot of current that the USB port isn't able to provide, and the Arduino ends up resetting or going into an unknown state (or you can try powering the servo from a separate power supply)
- The last thing I can think of is trying another Arduino/Teensy/ESP32 board to make sure there isn't a HW failure of some kind, but that will obviously depend on how many microcontrollers you have on hand

Shaorune777, 2024-08-10:
Hai.
I have same thing happend to me too.
What I use
Arduino mega
3 servos mg996r
own power supply for each servos

after move some bone in pose. is fine for a moment but after 20 second. the entire blender just lagging and freezing. the only way to stop this just disconnect button. and everything back to normal

while in lag state. the arduino just full bright TX led in the board

I try changing the volt to 5 to 6 volts. same happend.
changing mega to uno. still happened.
changing the USB cable. still same
put the usb cable from USB powerd hub to motherboard USB. still same. 

test in blender. 3.5.1
I tried in blender 4.2. the blender dont come with serial module.

dunno where is the problem

closed 2025-01-02 by knee-koh