knee-koh/MarIOnette #13: Can't get it to work (ESP32, 5 Stepper motors,blender4.1, ArduinoIDE2.3.2)

MrSunny29, 2024-07-12:
Hello,
I'm trying to animate a robot arm with 5 stepper motors.
I followed the whole tutorial and when I move it in Blender it should actually move for me too.
However, it doesn't always move and when it does, all the motors only go in one direction and not where they should.
I use an ESP32
Everything is wired correctly. If I control it manually using my own software it works.
I'm not sure if it's me, the chip or something else, as I'm still fairly inexperienced.
I hope someone can help me.
Good tutorial!
Thank you
MrSunny

knee-koh, 2024-08-23:
Hi, could you provide the stepper motors, motor drivers, and a picture of your MarIOnette actuator setup panel? A wiring diagram would also be helpful.

What could be happening  (depending on your ESP32) is some of the pins used for data transfer on the USB bus might be wired to the "dir' and "step" pins on the motor drivers, causing issues.

MrSunny29, 2024-09-10:
Hello,

I'm using **Nema 17 (39mm)** stepper motors, but I have motors from two different manufacturers. For one of them, I had to swap the two middle cables. I'm using **A4988 drivers**. I use a **esp32 devkit v1**.

The arm works perfectly when I send commands through the Serial Monitor in the Arduino IDE. I can give it coordinates and control it with joysticks.

[image]

The wiring is basicly like this but with 5 Drivers and Motors:

[image]

knee-koh, 2024-11-10:
Sorry for the delayed response!

Are you able to see the raw angle values change when you move the bones inside Blender? It's at the bottom of the Actuator setup tab (just where your screenshot cuts off).

MrSunny29, 2024-11-14:
I can see the raw angle value, and it changes when I move the robot. Beneath it is the output value.

knee-koh, 2024-11-16:
Hmmm, not sure what is causing these issues. Could you send me the .blend file so I can try running it here?

MrSunny29, 2024-11-17:
Is there a way to send it to you privately?

knee-koh, 2024-11-17:
You can send me a message through my [website contact form](https://www.nikovladimirov.com/about) with a link to the files. Please include the Arduino .ino and .h files as well.

closed 2025-05-07 by knee-koh