knee-koh/MarIOnette #14: On Blender Connect, Servos just rotate continuously

tommywright, 2024-07-15:
I purchased everything on the list, 3d printed the parts and assembled it all. The Arduino Nano is setup correctly on the breadboard and I've tested the servos with the Example sketch and they work fine.

In Blender, I'm using the pre-rigged scene as I'm already familiar with rigging. When I select my serial port and hit connect, the servos will just rotate counter clockwise continuously. 

I'd really love some help if possible.

tommywright, 2024-07-15:
I was on Blender 4.1 but I just tried Blender 3.6 and get the same thing.

tommywright, 2024-07-15:
Hmm... I hooked up a cheapo servo (SG90) and it seems to work fine! I'm not a servo expert but for some reason these MG 996Rs that were linked in the tutorial do not work.

knee-koh, 2024-07-15:
Hi, thanks for debugging and trying things out!

It's strange that the MG996Rs didn't work with the setup for the example. I know they are a little more power-hungry than the SG90s, but the AA pack should be enough to power everything (assuming there are 4 relatively new batteries loaded). If you have a power supply laying around, I would recommend you try running the robot from that to rule out power issues. If you still run into problems, then the servos Amazon/our supplier shipped you might be faulty.

closed 2024-07-15 by knee-koh

tommywright, 2024-07-16:
I read that the continuous servo will not work... that it has to be a 180 degree limited one. The link took me to a continuous MG996R servo. I ordered 2 new MG996R servos that are 180 degree.... https://www.amazon.com/dp/B07MFK266B

tommywright, 2024-07-16:
Well damn.. received those new servos and they don't work either. I'm now applying 5v to the servo through a Elegoo breadboard power converter. I guess it needs more than 5v.

knee-koh, 2024-07-16:
That sounds right, the Amazon page you listed shows 6.0v minimum with 7.4 recommended.

tommywright, 2024-07-17:
Okay.. I got it. It was the 5v board I was using... I guess it just didn't give it enough power. I'm linking the batteries directly to the breadboard now like you show in the video. He works but stops after a few seconds and I have to hit Disconnect and Connect again. Am I doing something wrong?