knee-koh/MarIOnette #15: support for more serial bus servos?

hcgilje, 2024-07-24:
Hi, I Iove this addon and hope you will continue to develop it.
I wonder if you can give some pointers on how to use other serial bus servos than the two you have added. 
I have a model from waveshare (identical to feetech): https://www.waveshare.com/st3020-servo.htm

I have used it with this serial bus controller: esp32 based, wifi and bluetooth, and it comes with an arduino library.
https://www.waveshare.com/servo-driver-with-esp32.htm
https://github.com/waveshare/Servo-Driver-with-ESP32

alternatively there is the TTlinker-mini board from feetech, but then you would need a microcontroller with two serial ports (or set up software serial which I guess is what you are doing with the LSServo?).

https://www.aliexpress.com/item/1005002425419341.html
The TTlinker works with the STS library: https://github.com/matthieuvigne/STS_servos

hcgilje, 2024-08-05:
I managed to get my waveshare servos working with marIOnette!
I used a teensy 4.1, the TTlinker board mentioned in my original post, and the SCservo library (found here: [Servo-Driver-with-ESP32/SCServo at main · waveshare/Servo-Driver-with-ESP32](https://github.com/waveshare/Servo-Driver-with-ESP32/tree/main/SCServo)
In blender I set the serial bus servo to Lewansoul.
In the generated config.h file I had to modify a couple of things to get it working with SCservo library:
line 13:  
`#define BUS_SERVO_BAUD 1000000`

line 45: 
```
// Bus Servos
// #include "LSServo.h"
// LSServo BusServos;

#include <SCServo.h>
SMS_STS BusServos;
```

line 143: 
```
// // Set Lewansoul servos
// if(motor_values[i][0] == 7){
// BusServos.SetPos(motor_values[i][1], motor_value, busServoSpeed); // Might need to remap the speed...
// }

// Set waveshare/feetech servos
if(motor_values[i][0] == 7){
BusServos.WritePosEx(motor_values[i][1], motor_value, busServoSpeed);
}
```

It would be great if waveshare/feetech could be added as a serial bus option.

the ST3020 servo has 12 bit resolution, how can I make full use of that with marIOnette?

Also, I am trying to understand how I can get the reading positions from the serial bus servos to blender. In the code there is a mode "R" but how do I set it?

knee-koh, 2024-08-23:
Thanks for looking into adding support for more actuators. Since there are so many different flavors of Bus Servos, it's hard to keep up with every version and have MarIOnette support each one. I also unfortunately don't have access to the ST3020, so I won't be able to update things until I'm able to test things properly myself.

As for 12-bit resolution, MarIOnette supports 16-bit resolution currently (it splits the values from Blender into two bytes and then re-combined them into a 16-bit integer on the Arduino). You should be covered by making sure you output 0-4096 to the Arduino in the MarIOnette panel.

For reading positions, I had some old legacy code from years ago that "sort-of" worked. I am currently in the process of re-writing that entire section in order to make it a little more user-friendly. Stay tuned for the next version, which should support reading values and creating animations through the hardware.

closed 2024-08-23 by knee-koh

hcgilje, 2024-08-25:
thanks for getting back to me. I was thinking more if it was possible to have a generic serial bus servo module where it would be possible to enter name of the servo library, constructor name and command for set position.
It is of course not a huge job for the user to tweak this in the generated arduino code.