knee-koh/MarIOnette #16: script to include animation in arduino code

hcgilje, 2024-08-25:
Would it be possible to have an alternative to saving the animation to a SD card, like an array or something similar?
The animation file size is so small it would easily fit on most microcontrollers.
I have looked at the option of converting the generated SD code to something that could be included in the arduino code, but this would be cumbersome to do for every animation.

knee-koh, 2024-11-12:
It is possible, though you will quickly run out of memory (especially on more basic microcontrollers with smaller PROGMEM). 

A 10 second animation, at 60fps with just 2 actuators comes out to 2.4KB (10sec x 60fps x 2motors x 2bytes per motor = 2400). I'll take a look at adding this in, but can't guarantee the auto-generated code will be able to keep track of multiple animations.

hcgilje, 2024-11-12:
ok, but newer esp32 controllers have flash memory of several megabytes. If I understand it right, code is written to flash memory?
Thanks for looking into this.

knee-koh, 2024-12-30:
I added support for this in the latest release. See the release notes for details.

Currently, only one animation is saved, and any new animations re-write everything before it. If you need more files stored, you'll have to copy the contents of the old animations.h file, generate a new animations.h file (with a new animation name), and then paste the old data into the new file. If I have more time, I'll work on getting the Python script to append to the file or overwrite if a given animation already exists.

closed 2024-12-30 by knee-koh