knee-koh/MarIOnette #19: I have not been able to get the SD card functionality working

trung-cyb, 2024-11-09:
**"I have not been able to get the SD card functionality working, even though I followed your instructions. Could you assist with this part?"**
Thank so much.

trung-cyb, 2024-11-11:
**"Thank you very much for your response.**
I am using Blender 3.6.17, and the MarIOnette version is attached in the image below. I am using an Arduino Nano and would like to play back the animation.
Currently, I can control the motor successfully. However, there is an issue with the SD Card playback. I have also enabled it in the Arduino IDE as per your description.
I also copied the 'test8' file after caching it to the SD Card. However, there is no motor playback activity when the Arduino is disconnected from MarIOnette in Blender.
I hope you can help. 
Thank you very much."
[image]

[image]
[image]

knee-koh, 2024-11-11:
Are you able to access the SD card with the Arduino independently (without MarIOnette). It looks like you have VCC hooked up to 5V instead of 3.3V (I've run into some adapters not liking this); can you follow this guide to see if the wiring and SD card are working: https://thesolaruniverse.wordpress.com/2022/09/01/external-sd-card-reader-arduino-nano-esp8266-and-esp32/?

When you open the test8 file in a text editor, do you see a header file followed by lots of lines of data? If it's just the header line, it means the Actuator panel in MarIOnette isn't being output. If everything looks like what I described, it's likely a wiring or code generation issue.

trung-cyb, 2024-11-11:
Thank you.
 It seems like the SD module is broken or has an issue. I have tried but couldn't read anything from the SD card. I will need to order another SD module to test it. 
 Thank you for your help.

trung-cyb, 2024-11-13:
> Are you able to access the SD card with the Arduino independently (without MarIOnette). It looks like you have VCC hooked up to 5V instead of 3.3V (I've run into some adapters not liking this); can [...]

**Thank you for your response**. I have tested and managed to connect to the SD card and run MarlOnette again. However, I don’t see the motors working according to the SD file through the test1 file.
Previously, I was able to control the motors using MarlOnette. I’m not sure if I made any mistakes.
[image]
[image]
[image]

I hope you can help me.

knee-koh, 2024-11-13:
Hmm, the file definitely looks like it has all the data written... Apologies for the lack of documentation on this feature, I'm working on getting a wiki set up that will hopefully expedite a lot of issues.

Did you change DEBUG_SD_PLAYBACK_TEST at the top of the .ino file from 0 to 1?

It looks like you're trying to trigger playback with P, which is the right character, but you need to give it the filename. Try "Ptest1" and see if either of those two solutions work.

trung-cyb, 2024-11-13:
*Dear Niko Vladimirov,*
My name is Huu Trung
After changing DEBUG_SD_PLAYBACK_TEST at the beginning of the .ino file
from 0 to 1, it worked—wonderful!

However, I noticed a few minor issues:

   - When I press the "P" key, it shows "File not found on SD card!" But
   when I press the reset button on the Arduino, it automatically replays the
   SD card.
   - This means that each time I want to replay, I have to press reset.
   - Additionally, the cache file in MarlOnette, when saved to a file,
   seems to play back the animation somewhat inaccurately.
   - I hope that if you ever create a guide on this topic, it would be
   fantastic.

Also, I’m sending you a product that uses your MarlOnette application for
you to see. How can I send it to you, or something similar, as a thank-you
gift? Thanks to you, I was able to complete this project.
https://drive.google.com/drive/folders/1E3RLo7Kj4LxOX9CDBd2SUEqJSyRjCTNN?usp=drive_link

I sincerely hope to have the opportunity to work with you and would like to
share the profits from selling the "ELBot" product with you in the future.

Thank you sincerely.

knee-koh, 2024-11-13:
Hi, I hid your previous comment because it had our names shown publicly in the Github thread (your email was hidden by github, so I couldn't reply directly).

You should be able to play animations through Serial commands by typing the letter P, followed by the name of the file to play. In your case, type "Ptest1" into the serial monitor, and it should play the file back. 

I'm not sure why the animation is not accurate. You can quickly check to see if the size of the file matches the animation from Blender: File size (bytes) = (number of motors) * 2 * (total animation frames). If the file size in the properties window on your computer is very different from this formula, something is getting lost while outputting the file from Blender. Otherwise, the microcontroller might be dropping frames while trying to play back the animation.

trung-cyb, 2024-11-14:
Hello,
Thank you for your response.
 There is an issue with controlling the stepper motor. Please take a look at the video I’ve attached. 
 https://drive.google.com/file/d/1HvxpRXMIXBU7ZE2TmdVyYYtS387JmdoQ/view?usp=sharing
 When connected to the Arduino, the motor automatically rotates a few turns without holding its position. This leads to an inability to control the motor's initial position. Have you encountered this issue before? I’ve tried adjusting the values, but it hasn’t worked, and the other stepper motors are showing the same behavior.

Thank you very much for your assistance.

knee-koh, 2024-11-16:
I have not seen this issue before. Are you using a Nano? I haven't used older Arduino boards in a while, and I believe the Nano shares a USB port for both debugging and serial. Sometimes connecting/reconnecting to the Serial port might reset the microcontroller and make it go through the setup sequence again (which could be why you are seeing the motor move).

closed 2025-05-07 by knee-koh