knee-koh/MarIOnette #20: the grid and servo demo: trying to understand how the LED system works

hcgilje, 2024-11-13:
There seem to be so many great features hidden in marIOnette, but which are a bit hard to grasp without documentation.
I would love to see a video explaining the setup for the LEDs in the blender grid and servo demo for instance.

knee-koh, 2024-11-13:
I'm working on getting a wiki hosted in the Github repo to explain all the features in more detail. That will cover how to modify the Geometry Nodes setup for individually addressable LEDs/strips/grids vs normal PWM LEDs. 

I just need time to add support for ESP32 and on-device animation storage first.

hcgilje, 2024-11-13:
sounds promising!

hcgilje, 2024-11-13:
..and let me know if you need help testing particular features