knee-koh/MarIOnette #21: no module named 'serial'

Sandystrong0, 2025-01-23:
After installing blender script i am not able to activate it in addons section and getting this error: 

[image]

knee-koh, 2025-01-23:
Hi, a couple of things can cause this:
- What operating system are you running?
- What Blender version are you using?
- What version of MarIOnette are you attempting to run?

Sandystrong0, 2025-01-24:
here,

i. windows 11
ii. blender 4.3.2 
iii. MarIOnette v1_0_6

are there any changes you might suggest that will work

knee-koh, 2025-01-24:
Try this: 
- Remove the add on from Blender
- Close Blender
- Launch Blender as administrator
- Reinstall the add on

stb1Cluny, 2025-01-25:
Same issue here.

Windows 10
Blender 4.2.1 LTS  & 4.3.2
MarIOnette v1_0_6

I did install the module before reading the installation instructions.
    -Uninstalled Blender and deleted the AppData
    -Reinstalled 4.3.2
    -Run as administrator
    -Same error

knee-koh, 2025-01-25:
Thanks for checking this, and sorry for the issue! It looks like one of the modules updated the way in which it checks which operating system is running in order to install the Serial module. It should now be fixed, please download the latest .zip file from here: [link](https://github.com/knee-koh/MarIOnette/tree/v1.0.7/Blender%20Script/Blender%20Version%204.2%20and%20Later/MarIOnette%20v1_0_6)

stb1Cluny, 2025-01-25:
Oh sure! I actually did get this working in Windows before I saw your reply. I followed the solution in Issue #3 

I had to loosen write permissions on the site-packages folder to get pip to install pyserial correctly. I was able to enable the add-on after that but I haven't tested to see if all the functionality is there.

I'll try to clean up my environment a bit and test your fix next.

stb1Cluny, 2025-01-25:
Amazing, confirmed your fix is working in a fresh installation of Blender 4.3.2 on Windows 10.

closed 2025-05-07 by knee-koh