knee-koh/MarIOnette #22: Lag/Latency?

Echolink50, 2025-01-26:
Thanks for the project. I have not tried it out yet but was wondering how much lag there is between blender motor movement? If is a consistent amount or does it vary? Can this be accurate in real time? Thanks

knee-koh, 2025-01-26:
Define "lag"... 

There's three components: 
- Speed of the Serial port transmission (this depends on how much data you're sending, your PC serial port + OS, and how the microcontroller handles Serial transmission)
- Arduino deciphering the message (this is mainly down to message size). Each actuator/LED roughly translates to a few microseconds each
- The actuator/LED itself. If you have a motor with a large geartrain, you won't get instant movement from the motor. LEDs typically respond near-instantaneously since they are simpler to drive.

I haven't measured the total latency time, but you can see the GIFs on the main page (and the teaser video) that most applications run in near real-time. I say "near real" because no system will be truly real-time due to the above points. For all of my applications it has been more than sufficient, and if Blender starts to lag due to a large number of actuators/LEDs, I just cache the animation locally to the microcontroller to prevent latency from Blender influencing motion.

Echolink50, 2025-01-28:
Ok. Without knowing all the tech language you used, I was considering using your addon for a virtual production setup. The physical camera motion and FPS would need to sync with the blender camera or the background wont look correct. If the lag/latency is consistent then a simple frame offset would work fine. I don't really understand the parameters of your system that are important for such a setup. Thanks for the information

closed 2025-05-07 by knee-koh